bambulab/BambuStudio #7470: Avoid Seam Placement on Overhang 避免在悬垂放置Z缝

hotellonely, 2025-07-04:
**Is your feature request related to a problem? Please describe.**
Placing seams on overhang would always create problems. It should be a rather easy task to avoid placing the seam on large overhangs.
自动生成的Z缝如果出现在悬垂上往往问题很多，一个比较简单的方案就是允许自动Z缝避开悬垂。

**Describe the solution you'd like**
Add an option to avoid placing seams on overhangs for automatic seam placements.
添加一个选项，让自动接缝算法可以避开悬垂。

**Describe alternatives you've considered**
When slicing, the slicer can scan the model to mark the overhangs as no-seam-zone. It's very similar to current "aligned" placing, but add a filter to choose the least overhang affected areas.
Aligned seam placement already has the information from previous and next layers. All one need to fulfill this useful function is to add the overhang angles into the package.
切片时对模型进行扫描，首先获得悬垂角度信息以后，像“对齐接缝”算法那样寻找合适的位置摆放接缝，只是这次额外添加一个针对悬垂角度的筛选。因为本来对齐接缝算法就已经有前后几层的信息了，额外加一个基于悬垂信息的判断应该非常容易。

hotellonely, 2025-07-08:
> 感谢您的反馈，我们正在内部对Z缝避开悬垂功能进行试验，请等待后续版本中的更新。

太棒了，真该开个给开发人员加鸡腿的打赏渠道lol

JohnSmithCYM, 2025-07-08:
@QingZhangBambu 客户要给你鸡腿:rocket:

QingZhangBambu, 2025-07-09:
haha, 鸡腿到我的碗里来

QingZhangBambu, 2025-07-21:
@hotellonely 新版本已经发了，现在调整的还比较保守。在多个条件下增加避让悬垂的因素，如果这个条件影响比其他的大就会避让悬垂

QingZhangBambu, 2025-07-21:
@hotellonely 欢迎使用之后提出后续改进意见

closed 2025-07-21 by QingZhangBambu