bambulab/BambuStudio #7475: Flow Rate Calibration does not work

Bax85, 2025-07-04:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Go to Calibration
2. Go to to Flow Rate
3. Go to Auto-Calibration
4. Choose nozzle, plate, filament and press Calibrate

### Actual results

Everything works as intended, calibration is printed and checked by micro lidar.
After 100% :
"Part of the calibration failed! It may help to clean the plate and retry. The failed test result will be deleted."
"Flow Ratio"
"Save Filament Preset"
"The failed test result has been dropped."

[image]

[image]

### Expected results

Your filament flow ratio is: x.xxx

### Project file & Debug log uploads

[debug_Fri_Jul_04_21_10_08_9696.log.zip](https://github.com/user-attachments/files/21066477/debug_Fri_Jul_04_21_10_08_9696.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Bax85, 2025-07-04:
Flow Rate Calibration works when downgrading the firmware to 01.08.05.00.

zhimin-zeng-bambulab, 2025-07-10:
It is because the calibration failed. You can try to clean the plate bed and recalibrate.

Bax85, 2025-07-14:
> It is because the calibration failed. You can try to clean the plate bed and recalibrate.

Opened a brand new smooth plate, changed filament color from Black Bambu PETG HF to Grey Bambu PETG HF.
Same issue, "Part of the calibration failed! It may help to clean the plate and retry. The failed test result wil be deleted."

So this can't be a hardware issue, this must be a software issue.

Earlier the error message was invalid tray and it wouldn't even start the print.
This new error message "Part of the calibration failed! It may help to clean the plate and retry. The failed test result wil be deleted." prints good on the plate but for some reason it does not read and save the scanned lidar results in bambu studio.

Calibration works very good with Bambu Studio version 01.08.05.00 and earlier versions.

zhimin-zeng-bambulab, 2025-07-14:
@Bax85 Could you share your  Studio log files and printer logs?

Or you can send it to my email separately.  My email address is  "zhimin.zeng@bambulab.com".  We promise that it will only be used for debugging and will be deleted immediately after use.