bambulab/BambuStudio #7477: Cloudfalre Verify Human Infinite Loop

arejaybee, 2025-07-05:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Built from source by yourself or others

### OS version

PopOS

### Printer

Bambu Lab P1S

### How to reproduce

    1. Launch Bambu Labs
    2. Click Online Models, Click MakerLab, or search anything from Home page
    3. The makerworld.com page is displayed asking the user to verify they are human
    4. 

https://github.com/user-attachments/assets/6dc975b5-3b5c-4e28-b093-f11e6519ac84

User clicks the Verify you are human checkbox

### Actual results

    1. The spinner loads for a while
    2. Then the box refreshes to verify you are human
    3. No matter how many times the user clicks this, they are never verified
    4. The application is genuinely unusable as this locks the user out of all features

### Expected results

The user is verified and navigates to the page

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21071597/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

arejaybee, 2025-07-05:
This issue presented itself a couple months ago - https://github.com/bambulab/BambuStudio/issues/6809
It was resolved for a short time, but the cloudflare validation seems to be broken again

coyoteran, 2025-07-05:
Started happening yesterday for me as well. Not a solution but the appimage works for me.

CLandsberg, 2025-07-07:
+1 with flatpack on endeavourOS

ant1fact, 2025-07-07:
+1 here

Bambu Studio version: 2.1.1.52 via Flatpak

OS:
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy

frank-pian, 2025-07-08:
Sorry, there was a firewall issue. Has it been resolved now?

jescanellas, 2025-07-08:
I still experience it, @frank-pian. 

Bambu Studio v2.1.1.52 via Flatpak

I'm on Linux Mint 22.1

rletendu, 2025-07-11:
Looks like this is now working again starting from yesterday, same BambuSutdio 2.1.1.52 flat pack  and Linux Mint 22.1 ...