bambulab/BambuStudio #7478: Auto Flow Rate Calibration not working

Jfurry0918, 2025-07-05:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 24H2

### Additional system information

12th Gen Intel(R) Core(TM) i7-12700KF   3.60 GHz
32.0 GB RAM
RTX 3070Ti

### Printer

Bambu Lab X1C

### How to reproduce

Go to Calibration
Go to to Flow Rate
Go to Auto-Calibration
Choose nozzle, plate, filament and press Calibrate (.4mm)

### Actual results

Auto calibration will fall with error message of "Selected diameter and machine diameter do not match"
Selected diameter in printer settings and all slicer settings are set the same at .4mm

[image]

[image]

[image]

[image]

### Expected results

Auto calibration would start and do the calibration.

Manual calibration works perfectly fine without error.

Auto Flow Dynamics works without issue.  Only issue is with Auto Flow Rate Calibration

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21071841/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sergei-b, 2025-07-18:
> Please update the latest X1C firmware to see if this problem still exists.

This problem still exists with 01.09.01.00 firmware which is the latest version and Bambu Studio 2.1.1.52 which is the latest Linux version

zhimin-zeng-bambulab, 2025-07-18:
Please wait patiently for the next firmware update version