bambulab/BambuStudio #7489: Cannot build on Manjaro due to cmake_minimum_required() being too low.

niel, 2025-07-06:
### Bambu Studio Version

n/a

### Where is the application from?

Built from source by yourself or others

### OS version

Manjaro

### Printer

A1

### How to reproduce

on a Manjaro Linux distro clone the repo and run BuildLinux.sh

### Actual results

attempting to build using BuildLinux.sh result in failure with the dependencies because the cmake_minimum_required() version is too  low.   The version of Cmake on manjaro will not work with the version set lower that 3.5.  It is recommended to set the version to 3.5...4.0 to indicate if the cmake script support new standards.

### Expected results

project should build without errors.

### Project file & Debug log uploads

there is no project file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2025-07-28:
We can only guarantee Linux build support on Ubuntu 22 and Ubuntu 24. For other systems, you will need to resolve issues on your own. Thank you.

closed 2025-07-28 by MackBambu

niel, 2025-07-29:
I think you have missed the point on this. It is a build issue because the newer versions of CMake will refuse to build when the build files indicate a version of Cmake before 3.5, as your build files do. This problem will affect all OSes that have a newer Cmake version, including Ubuntu 24.04.2.

Ubuntu also has other problems with building, which I have opened a separate issue for.