bambulab/BambuStudio #7495: Flow Calibration option missing on A1 Mini

Stagecoachsi, 2025-07-06:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Printer

Bambu Lab A1 mini

### How to reproduce

Al mini with [.2mm] nozzle, any setting will not let you perform a Flow calibration before sending the print. 

[image]

### Actual results

Al mini with [.2mm] nozzle, any setting will not let you perform a Flow calibration before sending the print. 

### Expected results

Al mini with [.2mm] nozzle, Should let you perform a Flow calibration before sending the print. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21090919/log.zip)
[B14R Bottom Plate v5.zip](https://github.com/user-attachments/files/21090918/B14R.Bottom.Plate.v5.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

fahaaaxiki, 2025-07-21:
Please upload log through Handy APP, and give us your ticket number

Upload log files through Bambu Handy
Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
To upload files, go to Me → Support Tickets → Choose the ticket for log upload and proceed with the operation.
app_support_ticket

[image]

closed 2025-07-21 by fahaaaxiki

aschubert-mathema, 2025-08-14:
Still an issue with newest version 2.2.0.85 on A1 Mini with 0.4mm nozzle on mac os. Reverting back again.

fahaaaxiki, 2025-08-14:
> Still an issue with newest version 2.2.0.85 on A1 Mini with 0.4mm nozzle on mac os. Reverting back again.

May this problem be solved after returning to the previous Studio version?

Stagecoachsi, 2025-08-15:
Still experiencing this issue.

Darius-bd, 2025-08-17:
I'm having the issue as well, there's no option, but the printer does a Dynamic Flow Calibration before every print and I have no way to stop that from happening. Printer firmware 01.03.01.02

aschubert, 2025-09-28:
> We have fixed the similar problem of H2S in V2.2.2.56，please try again.

Seems also to have fixed it for me. A1 Mini latest firmware and V2.2.2.56 on Mac OS show Flow Calibration again. Thx!

brummbrum, 2026-10-01:
I reported this problem in the Orca repository, see https://github.com/OrcaSlicer/OrcaSlicer/issues/16033

It may be possible to get around this by updating the A1 family printer firmware as newer firmware did introduce additional capability reporting flags, however the newer firmware versions come with massive downsides, ie locking down the printer in many ways.

@XinZhangBambu 
A straight forward simple fix without killing it for other Bambu printers is to add a logical OR path that preserves the older logic:
```
is_support_pa_calibration =
    jj["support_flow_calibration"].get<bool>();

+ is_support_flow_calibration =
+   jj["support_flow_calibration"].get<bool>();
```
and later

```
- m_checkbox_list["flow_cali"]->Show(obj->is_support_pa_calibration);
+ m_checkbox_list["flow_cali"]->Show(
    obj->is_support_pa_calibration || obj->is_support_flow_calibration
);
```
Of course one could argue that the change could be made simpler since the two boolean variables are the same. But it is effective without messing with all the other changes from the newer capability flags (namely the evaluation of the bits in home_flag and fun). In short: this simple change will restore the correct capability mapping for printers using older firmware and respects the nomenclature changes you introduced over time.