bambulab/BambuStudio #7496: Strong Tree Supports printed on air over holes in part

240Robert, 2025-07-06:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

H2D

### How to reproduce

[Bambu Studio Bug - heavy tree supports created in air.pdf](https://github.com/user-attachments/files/21091129/Bambu.Studio.Bug.-.heavy.tree.supports.created.in.air.pdf)

### Actual results

When I switch to heavy tree supports, Bambu Studio creates them, but some of the supports that go over holes in my part are printed as zigzag lines in air (these would fall into the hole if I printed the file). This is currently preventing me from using the heavy tree supports.

### Expected results

I would like the tree supports to generate properly and be supported over holes. Please let me know if there is anything I can do to force the program to create the supports properly.

### Project file & Debug log uploads

[RI-220x220x120_2025-06-29.zip](https://github.com/user-attachments/files/21091152/RI-220x220x120_2025-06-29.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

NothFie, 2025-07-08:
@240Robert Thanks for your feedback. When there are too many nodes per layer, the nodes will grow in SQUARE to reduce calculation so it looks like "zig-zag". When nodes meet the holes, the center of their square are falling on the model, so they disappear suddenly and leave some lines floating. I think these floating lines will have an impact, but not a significant one (you could treat them as bridges). If you want to get better result, I recommend Organic Tree.

NothFie, 2025-07-11:
@240Robert 
1. Taking this model as an example, Organic Tree might have some THIN and TALL branches to reach the buildplate. These branches may become unstable and wobble when printed at greater heights. Unfortunately, it's hard to avoid this situation by changing support params, so for this kind of models, we recommend Hybrid Tree.
2. Setting wall_count to 2 is effective in most situations.
3. Make sure your build plate has sufficient adhesion.
4. Slightly reduce the parameter "Support-Branch angle" will reduce the risk of support warping.