bambulab/BambuStudio #7497: Bambu studio home page does not load

SuperFatty, 2025-07-07:
### Bambu Studio Version

02.01.01.52

### Where is the application from?

Bambu Lab Official website

### OS version

microsoft 11

### Additional system information

         Operating System: Windows 11 Home 64-bit (10.0, Build 22631) (22621.ni_release.220506-1250)
                 Language: English (Regional Setting: English)
      System Manufacturer: Micro-Star International Co., Ltd.
             System Model: MS-7D93
                     BIOS: 1.B1 (type: UEFI)
                Processor: 13th Gen Intel(R) Core(TM) i9-13900K (32 CPUs), ~3.0GHz
                   Memory: 65536MB RAM
      Available OS Memory: 65228MB RAM
                Page File: 19771MB used, 49552MB available
              Windows Dir: C:\Windows
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
         User DPI Setting: 96 DPI (100 percent)
       System DPI Setting: 96 DPI (100 percent)
          DWM DPI Scaling: Disabled
                 Miracast: Available, no HDCP
Microsoft Graphics Hybrid: Supported
 DirectX Database Version: 1.7.2
           DxDiag Version: 10.00.22621.3527 64bit Unicode
Card name: NVIDIA GeForce RTX 5070 Ti

### Printer

NA

### How to reproduce

Open bambu studio, 
home page never loads, just white screen

### Actual results

I can still do everything, just cant use anything on the home screen, so I can't log in.

### Expected results

maker studio should load and I should be able to log in

### Project file & Debug log uploads

[project.zip](https:/

[debug_network_Sun_Jul_06_18_36_03.log.zip](https://github.com/user-attachments/files/21093813/debug_network_Sun_Jul_06_18_36_03.log.zip)

/github.com/user-attachments/files/21093811/project.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

MackBambu, 2025-07-28:
Can you try switching to a different version of WebView2? It might improve the issue. We've detected that Microsoft's WebView2 sometimes encounters errors, and although the problem has been reported upstream, it hasn't been resolved yet.
https://developer.microsoft.com/zh-cn/microsoft-edge/webview2?form=MA13LH
https://github.com/MicrosoftEdge/WebView2Feedback/issues/5220?open_in_browser=true#event-17428098527

SuperFatty, 2025-07-28:
> MicrosoftEdge/WebView2Feedback#5220

Turns out it was a Windows thing. I forced an update on Windows and it went
through just fine.

JohnSmithCYM, 2025-08-14:
Can you reproduce this problem with https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85? If not, please tell us.

closed 2025-08-19 by JohnSmithCYM