bambulab/BambuStudio #7499: Bambu CLI :  --curr-bed-type param seems to be ignored

arirangz, 2025-07-07:
### Bambu Studio Version

2.0.4.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 pro

### Printer

Bambu A1 Mini

### How to reproduce

1. In CLI, generate a new 3MF using the param --curr-bed-type="Textured PEI Plate"
2. Open the 3MF
3. You will notice the selected plate is the on that was previously selected before closing bambu

### Actual results

You will notice the selected plate is the on that was previously selected before closing bambu

### Expected results

Bambu Studio should select the plate according to the param --curr-bed-type

I checked the source code and the curr bed type param is there:
https://github.com/bambulab/BambuStudio/blob/fcd50188ad6d6d23adac265a1fa3f234cecd1c84/src/libslic3r/PrintConfig.cpp#L843

Thanks

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-07-16:
Hello @arirangz 

please show us the detailed command on the CLI
thanks

arirangz, 2025-07-16:
Hello @lanewei120,

Here an example:
`"C:\Program Files\Bambu Studio\bambu-studio.exe" "C:\bambu\test.stl" --export-3mf "C:\bambu\test4.3mf" --debug=2 --load-filaments "C:\bambu\tpu.json" --load-settings "C:\bambu\process.json;C:\bambu\machine.json" --slice 0 --curr-bed-type="Textured PEI Plate"`

The 3MF file is correct but the selected plate is not Textured PEI Plate but the previously selected plate last time I open Bambu Studio.

So it seem this parameter is ignored.

Thanks

lanewei120, 2025-07-16:
this is not the CLI related issue, but the behavior of bambu studio
you can turn off this feature in the preference window

[image]

arirangz, 2025-07-16:
Thank you very much @lanewei120, it's working good now.

So on what "auto plate" is based on?

Thanks

arirangz, 2025-07-16:
Ok I see, thanks for the explanation!

closed 2025-07-17 by lanewei120