bambulab/BambuStudio #7505: Need to resync filaments every time

chrmenne, 2025-07-07:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Start Bambu Labs Studio. Or have it started by selecting "3D-Print" in Fusion.

### Actual results

The list of filaments is a mess and requires a manual re-sync with the following dialog every time.

### Expected results

There should be a way to at least remember the last filament list used.

### Project file & Debug log uploads

Not needed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-07-08:
@chrmenne Thank you for your feedback. Could you please provide a screenshot? Every time we open the project file, the consumable list is displayed according to the consumables contained in the project file.

chrmenne, 2025-07-08:
@BoBoYu-BamBu I'm puzzled. I set it up the way I want it to be, in order to give you a screenshot of what I need, to compare with a screenshot of what I get. But for some reason, it was now *mostly_* persisted. This is the screenshot of my actual setup:

[image]

This is what I get now on restart:

[image]

It used to be much worse, I don't know what has changed or why - the 3DK PLA IR was all over the list and instead of the @custom presets, the stock presets were used. Right now it's only the "PLA-CF" preset that gets reverted to "3DK PLA IR" whenever I start Bambu Studio by selecting "3d-Print" in Autodesk Fusion via a 3mf file. Much better now, but still not perfect. I don't know which project file Bambu Studio is loading when it's started like that., I assume there is some kind of default?

chrmenne, 2025-07-19:
[image]

Now it's getting wild again.

Haidiye00, 2025-11-20:
https://github.com/bambulab/BambuStudio/releases/tag/v02.04.00.70
Hello, the new version has been released, please update.