bambulab/BambuStudio #7509: Cannot clear search terms in home screen

Crossbow78, 2025-07-07:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1

### How to reproduce

1. Go to Home screen in BambuStudio
2. Enter any search term "abc" and press Enter
    > Online Models are shown, filtered by the search term "abc"
3. Clear the search box (by clicking the x, or manually)
    > Nothing happens
4. Press Enter in the now empty search box to trigger a blank search
    > Nothing happens
5. Navigate to any section (eg. Home, or Recently Opened, or change sort on Online Models)
    > The previous search term is reinstated

### Actual results

It is impossible to go back to browsing all models after having done a single search.

### Expected results

One of the below:
* Clicking the x icon in the search box should immediately have effect and show all models
* Pressing Enter in an empty search box should force a refresh and show all models
* Navigating to another section after clearing the search box should not reinstate the previous search term

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-07-08:
@Crossbow78 Thank you for your feedback. Please submit a work order on the mall website based on the screenshot provided. The mall team will follow up on this issue afterwards.this page is embedded within the studio software in the mall. Therefore, all issues related to this page will be handled by the mall side. I have already informed the mall team and they will make the necessary changes. Thank you.     Address：https://makerworld.com/en/3d-models
[image]

paxel, 2025-07-10:
I was also struggeling with that. workaround is to just search for a single blank.

IIRC

closed 2025-07-10 by Crossbow78