bambulab/BambuStudio #7511: Unlock Extruder Clearance settings in Bambu Studio

slothy89, 2025-07-08:
**Is your feature request related to a problem? Please describe.**
- In the past users were able to manually define the exruder clearance radius for per-object printing, this is now locked out.

**Describe the solution you'd like**
- Make these settings available for users to modify - Even if it is only hidden behind the "Advanced" view
- This would allow reducing the calculated clearance when models permit.

**Describe alternatives you've considered**
- Orca Slicer

**Additional context**
- By allowing manual definition of this setting for Bambu printers, you can tune the clearances required for printing by-object prints. For instance, the max value set is based on either the rear of the toolhead, or the Lidar on the X1.
- On the P1, the Lidar doesn't exist, so the clearance should already be reduced to 57mm from 68.
- If organised intellegently, you can reducce the required clearances significantly due to the nozzle being off centre from the toolhead.

I often print smaller batch items with clearances as low as 34mm allowing many more copies to be placed on the bed per batch. By ordering the object printing from left to right, front to back, the clearances of the left side and front of the toolhead are the only ones that matter.

Being forced to have double that is a huge issue for productive printing.

PixelyIon, 2025-07-15:
+1 - The default setting seems to be too conservative, and this is limiting for advanced users.

JWA-0030, 2025-12-07:
+1,  way too conservative. Let the advanced users edit this config

leuys, 2025-12-16:
+1, yes please. OP has described the issue perfectly. There is a lot of potential for batch printing by-object once clearance (and print order) can be manually set. Have a warning message appear if you must, indicating potential extruder crashes when default value is altered.

cjnaz, 2026-05-02:
+1.  This is a serious and unnecessary restriction for batch printing / bulk production.  Also an easy fix - enable the GUI Extruder Clearance Max Radius box for edit.

Since the GUI wont let me change the setting, I edited C:Users\<me>\AppData\Roaming\BambuStudio\system\BBL\machine\fdm_bbl_3dp_001_common.json
    "___extruder_clearance_max_radius": "68",
    "extruder_clearance_max_radius": "25",

This specifies about one inch from nozzle to the head left side and front.  Print sequence "By object" layout must be from left to right then front to back. This works fine for shorter objects.  Watch out for the rod clearance.  I can get 25 1in diameter rings on a plate.

RELATED PROBLEMS
1) The Auto Arrange tool is not useful since it does not consider print order/sequence.  One must lay them out  manually.

I'm taking the responsibility to lay out the spacing and order of printed objects.  Note this is on a P1S.