bambulab/BambuStudio #7520: Send Print Job tab error

Sandisoliandi, 2025-07-08:
### Bambu Studio Version

v02.01.01.52

### Where is the application from?

Bambu Lab Official website

### OS version

8.1

### Additional system information

intel(R) core(TM)i7-3770 CPU @3.40GHz-3.90GHz

[image]

[image]

8gb ram
64bit operating system

### Printer

p1s

### How to reproduce

Send Print Job tab error

[image]

### Actual results

Send Print Job tab error

### Expected results

Send Print Job tab error

### Project file & Debug log uploads

Send Print Job tab error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBambu, 2025-08-27:
Thank you for your feedback. We have developed a software version to address your issue; could you please verify if it works? If it does, we will merge it into the release version.
https://github.com/XinZhangBambu/BambuStudio_bug_fix/actions/runs/17255536799#artifacts

[image]

BoBoYu-BamBu, 2025-12-02:
Thank you for your feedback. This issue has been fixed. Please update to the latest version of Studio for verification. Download link: https://bambulab.cn/zh-cn/download/studio