bambulab/BambuStudio #7527: 内存卡兼容问题

xieenman, 2025-07-09:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

win10-2009

### Printer

P1S

### How to reproduce

使用大于64G内存卡，在打印机内执行格式化操作后，打印机内可正常存取文件，可以正常，可以正常打印、查看缓存的模型文件。使用读卡器将内存卡连接到电脑，也可以正常存取文件。但是BambuStudio查看设备时提示储存异常。延时摄影、模型均显示为空。

### Actual results

空

### Expected results

 空

### Project file & Debug log uploads

 空

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-07-09:
@xieenman 感谢您的反馈，能提供一下截图和log文件吗？

xieenman, 2025-07-10:
满满
***@***.***

 

 

从QQ邮箱发来的超大附件

BambuStudio截图和打印机log.zip (28.60M, 无限期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=66343131d0430e981115157b1e330a494d40070759010953180057045d1e0a0550511c550b065b4b030003025a075a5400570801382d3824545953446b474d025c5b8ce9f58f82ab81c6e29083c95409521a4b58483305&t=exs_ftn_download&code=5411838f