bambulab/BambuStudio #7528: libOSMesa.so.8: cannot open shared object file: No such file or directory

Karmicatosaurus, 2025-07-09:
### Bambu Studio Version

02.01.01.52

### Where is the application from?

Bambu Lab Official website

### OS version

Fedora 42

### Printer

Bambu Lab A1 + AMS Lite

### How to reproduce

Just try to launch the software

### Actual results

**********@***********:~/3D$ ./Bambu_Studio_linux_fedora-v02.01.01.52.AppImage 
/tmp/.mount_Bambu_i6ysOA/bin/bambu-studio: error while loading shared libraries: libOSMesa.so.8: cannot open shared object file: No such file or directory

The new version of mesa is 25.1.4-2. Since this version, the libOSMesa.so. 8 library is no longer available.

I am obliged to reinstall mesa-libOSMesa.x86_64 (dnf install mesa-libOSMesa.x86_64)
 and to block the version to regain access to Bambu Lab Studio (dnf versionlock add mesa-libOSMesa-0:25.0.2-3.fc42.x86_64)

### Expected results

the software crashes

### Project file & Debug log uploads

No log generated by the software, just crash.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bedfelllow, 2025-07-11:
Confirmed on Fedora 42.

colincross, 2025-07-19:
If you're using Fedora 42 you can install `mesa-compat-libOSMesa` instead of `mesa-libOSMesa` to avoid downgrading the rest of the mesa libraries:
```
sudo dnf install mesa-compat-libOSMesa
```

asm0dey, 2025-07-20:
Same on the new Arch

Karmicatosaurus, 2025-07-20:
> If you're using Fedora 42 you can install `mesa-compat-libOSMesa` instead of `mesa-libOSMesa` to avoid downgrading the rest of the mesa libraries:
> 
> ```
> sudo dnf install mesa-compat-libOSMesa
> ```

It works but it would be good if the users do not have to do this manipulation.

asm0dey, 2025-07-21:
Don't have such a package on Arch :(

MackBambu, 2025-07-29:
same #6951