bambulab/BambuStudio #7530: X1C The Print Again function from the LCD does not use the Auto-refill on the AMS.

gary-justausername, 2025-07-09:
### Bambu Studio Version

latest

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Additional system information

X1C. The Print Again function from the LCD does not use the Auto-refill on the AMS.

### Printer

X1C

### How to reproduce

Select the Print Again function from the LCD until the spool runs out

### Actual results

An error

### Expected results

To continue with the auto-refill slot

### Project file & Debug log uploads

No need

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-07-10:
@gary-justausername  Thank you for your feedback. May I ask if you have two identical sets of consumables? Please check if the option "AMS Settings" under the "Device" page is selected?
[image]

gary-justausername, 2025-07-11:
Hi,,
Yes I have checked on the printer and in Studio. Only happens when "print
again" on printer.