bambulab/BambuStudio #7531: Bambu studio se cierra inesperadamente

56-Tomas-56, 2025-07-09:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz
RAM: 32GB
Graphics: NVIDIA Quadro RTX4000

### Printer

Bambu Lab P1S

### How to reproduce

Cuando lamino una placa e intento visualizarla, el programa se cierra sin dar errores. Tampoco me permite imprimir la placa.

### Actual results

No puedo ver la placa laminada ni imprimirla

### Expected results

Que se muestre la placa laminada y poder enviarla a la impresora.

### Project file & Debug log uploads

Project: https://makerworld.com/es/models/1023490-domino-set-with-box?from=search#profileId-1005291

[log.zip](https://github.com/user-attachments/files/21148374/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-07-10:
@56-Tomas-56 Gracias por su comentario. ¿Puede proporcionar capturas de pantalla para describir su problema en detalle?

56-Tomas-56, 2025-07-10:
@BoBoYu-BamBu Adjunto un documento pdf con diferentes capturas del proceso y una breve descripción. Adjunto también los archivos log creados.
Muchas gracias.

[Capturas.pdf](https://github.com/user-attachments/files/21166890/Capturas.pdf)

[log1.zip](https://github.com/user-attachments/files/21166936/log1.zip)

56-Tomas-56, 2025-07-14:
@guanyun-gudujian 
Muchas gracias. Parece que con este paquete si que funciona el proyecto. ¿Mejor que utilice siempre los archivos que me has enviado?