bambulab/BambuStudio #7537: Device tab crashes while printing job is active

juaninachon, 2025-07-10:
### Bambu Studio Version

Flatpak 2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Ubuntu 24.04

### Additional system information

# System Details Report
---

## Report details
- **Date generated:**                              2025-07-10 09:10:58

## Hardware Information:
- **Hardware Model:**                              Gigabyte Technology Co., Ltd. B250M-DS3H
- **Memory:**                                      48.0 GiB
- **Processor:**                                   Intel® Core™ i7-7700 × 8
- **Graphics:**                                    NVIDIA GeForce RTX™ 3060
- **Disk Capacity:**                               3.0 TB

## Software Information:
- **Firmware Version:**                            F6
- **OS Name:**                                     Ubuntu 24.04.2 LTS
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            X11
- **Kernel Version:**                              Linux 6.11.0-29-generic

### Printer

A1 mini

### How to reproduce

1. Print plate
2. Job is send to printer and executes properly, but BambuStudio exits (i.e., crashes)

### Actual results

3. Reopen app
4. Click on device tab
5. Displays expected content (e.g., printing job in progress) but crashes again after ~1 second

### Expected results

BambuStudio should not close suddenly, but allow navigation to device tab and initialization of camera livestream

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21162700/log.zip)

[Grimreaper with blade.zip](https://github.com/user-attachments/files/21162721/Grimreaper.with.blade.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

XinZhangBBL, 2025-07-11:
Halo. We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

daniel-mooney, 2025-07-27:
I had the exact same issue with BambuStudio crashing during a print after I select the device tab. I am also running Ubuntu 24.04.

I followed the GDB procedure, and the app faired even worse. I encountered a seg fault on start. My gdb log files are attached.

[bambu_gdb_log.txt](https://github.com/user-attachments/files/21451027/bambu_gdb_log.txt)

daniel-mooney, 2025-07-29:
> Could you reproduce the problem to offer log for us. you can refer to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log. Thank you very much for your cooperation.

Here are some of the log files:

[debug_Sun_Jul_27_13_43_16_2.log.0.txt](https://github.com/user-attachments/files/21478425/debug_Sun_Jul_27_13_43_16_2.log.0.txt)
[debug_Sun_Jul_27_12_50_40_2.log.0.txt](https://github.com/user-attachments/files/21478426/debug_Sun_Jul_27_12_50_40_2.log.0.txt)
[debug_Sun_Jul_27_12_48_30_668.log.0.txt](https://github.com/user-attachments/files/21478424/debug_Sun_Jul_27_12_48_30_668.log.0.txt)

XinZhangBBL, 2025-07-29:
@sonicer105  The issue has been resolved.  please update to the latest version https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60.

> I'm also having this issue:
> 
> ```
> Thread 107 "bambustu_main" received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0x7ffd5bf596c0 (LWP 361)]
> 0x0000555557cb5d7a in boost::thread::interruption_requested() const ()
> (gdb) bt full
> #0  0x0000555557cb5d7a in boost::thread::interruption_requested() const ()
> #1  0x0000555557244cd4 in Slic3r::MachineObject::update_slice_info(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int)::{lambda()#1}::operator()() const ()
> #2  0x0000555557cb5b75 in thread_proxy ()
> #3  0x00007fffec499261 in start_thread (arg=<optimized out>) at pthread_create.c:447
>         ret = <optimized out>
>         pd = <optimized out>
>         out = <optimized out>
>         unwind_buf = {cancel_jmp_buf = {{jmp_buf = {32, 2788365830711201636, 140726146275008, -51640, 0, 140737488291792, 2788365830723784548, 2789747464291798884}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
>         not_first_call = <optimized out>
> #4  0x00007fffec51dd14 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100
> ```
> 
> Full log: [bambu_gdb.log](https://github.com/user-attachments/files/21478282/bambu_gdb.log)

sonicer105, 2025-07-29:
> [@sonicer105](https://github.com/sonicer105) The issue has been resolved. please update to the latest version https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60.

Maybe a stupid question, but how do I do that? flatpak tells me there's nothing to do on update.

```
linuxpony@Sailex-UPC5:~$ flatpak info com.bambulab.BambuStudio

Bambu Studio - Get perfect 3D prints!

          ID: com.bambulab.BambuStudio
         Ref: app/com.bambulab.BambuStudio/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 2.1.1 Public Release
     License: AGPL-3.0-only
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 701.5 MB
     Runtime: org.gnome.Platform/x86_64/48
         Sdk: org.gnome.Sdk/x86_64/48

      Commit: 89ea13aa4a965cfd6a5538503f905b16edc24a416e94d88f68cef734e7504807
      Parent: 62ca07241b7e88515dca0bdde3bc72d05fd7ff541dbb3d76205761ce5a4b244c
     Subject: Bump runtime version to 48 (3371e1dbc9de)
        Date: 2025-07-05 20:58:58 +0000
linuxpony@Sailex-UPC5:~$ flatpak update com.bambulab.BambuStudio
Looking for updates…

Nothing to do.
```

XinZhangBBL, 2025-07-29:
@sonicer105  Flatpak versions are made by other enthusiasts. After checking, there's no beta version of our software available. If you need it urgently, you can use the AppImage files. We will also release the corresponding official version of the software later.

> [@XinZhangBBL](https://github.com/XinZhangBBL) Those are appImage files. How do I get the flatpak or do I have to wait?

sonicer105, 2025-07-29:
oh okay. I'll go poke the flathub package maintainer then.

hadess, 2025-07-29:
> Builds of beta versions are usually available in a pull request.

In this case: https://github.com/flathub/com.bambulab.BambuStudio/pull/64

XinZhangBambu, 2025-09-24:
@sonicer105 The fixed version has been published. https://github.com/bambulab/BambuStudio/releases/tag/v02.02.02.56

closed 2025-12-02 by BoBoYu-BamBu