bambulab/BambuStudio #7553: Crash during G code build

MattTowers, 2025-07-12:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Device Name	Matt-DSL
Processor	12th Gen Intel(R) Core(TM) i9-12900HX   2.30 GHz
Installed RAM	64.0 GB (63.6 GB usable)
Graphics Card	NVIDIA RTX A4500 Laptop GPU (16 GB), Intel(R) UHD Graphics (128 MB)
Device ID	1E4088ED-E906-4DB7-97B9-87ACC430D608
Product ID	00355-62664-75332-AAOEM
System Type	64-bit operating system, x64-based processor

### Printer

Bambu H2D

### How to reproduce

1. Select preview tab after prepare. 
2. Gets to G code build and crashes

### Actual results

Application closes

### Expected results

Bambu studio closes

### Project file & Debug log uploads

[CC_01.zip](https://github.com/user-attachments/files/21192925/CC_01.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

rouyuanzi-bambu, 2025-07-18:
Hello, thank you for your feedback. To help us further investigate the issue, could you please let us know how the model was obtained?
In the meantime, we recommend importing the original model file (such as STL or STEP format) for slicing, and then saving it as a 3MF file before trying again.
Additionally, we’ve released the latest Beta version. You can download and install it via the following link. Please try using this version with the original model file to see if the issue persists: https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60

rouyuanzi-bambu, 2025-07-18:
We sincerely apologize for the inconvenience caused. The model can be sliced and printed normally on our test. Based on our preliminary investigation, the issue may be related to your graphics card.We recommend that you follow the troubleshooting steps provided in the following link: 
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing
If the problem persists, please follow the steps below to provide us with the Beta version log files so that we can further investigate and assist you in resolving the issue. Thank you for your understanding and support!
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

ShanCang-BamBu, 2025-08-05:
Thank you very much for your feedback!
Wishing you endless joy in the world of 3D printing!

closed 2025-08-12 by zhimin-zeng-bambulab