bambulab/BambuStudio #7566: Bambu Studios Always crashes sending gcode to Voron

pedrof, 2025-07-14:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

CPU: Intel(R) Core(TM) i7-14700K   3.40 GHz
RAM: 64.0 GB (63.7 GB usable)
Video Card: Nvidia GeForce RTX  3080 10Gb

### Printer

Voron Trident 350mm and Voron 2.4 350mm

### How to reproduce

1. Launch Bambu Studio
The program opens without issue.

You can navigate menus, open STL files, and slice normally.

2. Load a Model
You import or drag-and-drop a model (e.g., .stl or .3mf) into the workspace.

3. Select Voron printer 

4. Select Slice Plate, Slicing proceeds as expected — no crash at this point.

5. Prepare to Send the Print
You click on "Print" and select Upload or Print 

The send window for sending the gcode appears but the bars never show progress. 

❌ Bambu Studio instantly crashes and closes.

### Actual results

The Bambu Studios Crashes after sending gcode thru network. 

### Expected results

The expected results are to send the gcode to the 3d printer, as it used to on older versions. 

### Project file & Debug log uploads

[Pliers Holder.zip](https://github.com/user-attachments/files/21207487/Pliers.Holder.zip)

[log.zip](https://github.com/user-attachments/files/21207488/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

pedrof, 2025-07-22:
> Halo. We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

I already attached a log and dump file with the issue .Sent from my iPhone

XinZhangBBL, 2025-07-23:
@pedrof 
We are very sorry for the inconvenience caused to you. The stack of the crash dump failed to be parsed because the symbol server lacks relevant information. Could you please upgrade to the latest version to check if the problem still exists and upload the latest dump file?