bambulab/BambuStudio #7568: Delete saved models in collection

gary8688, 2025-07-14:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

x1c

### How to reproduce

click on model
select delete
confirm delete
nothing happens

### Actual results

"delete" keeps repeating but nothing else happens

### Expected results

model should be deleted from collection

### Project file & Debug log uploads

[debug_network_Sun_Jul_13_17_53_09.log.zip](https://github.com/user-attachments/files/21208890/debug_network_Sun_Jul_13_17_53_09.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gary8688, 2025-07-18:
> Hello, thank you for your feedback. We have attempted to reproduce the
> issue based on the steps you described but were unable to do so. To help us
> investigate further, could you please provide the model project file that
> cannot be deleted, along with a video showing the operation? Thank you for
> your cooperation and support!

20250718_054310.zip
<https://drive.google.com/file/d/1vzckzEgg-fDyjcuSjN-YJvevHSRwm2cF/view?usp=drive_web>
I hope this video comes through to you and it helps you to resolve my issue
with deleting models from my collections. Thanks

rouyuanzi-bambu, 2025-07-21:
Hello, it shows that I don't have permission to open the file on my end.I have already sent a request for access.
If it's convenient for you.can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1

gary8688, 2025-07-29:
> Hello, please provide a reproducible GIF again. Thank you, it will be
> convenient for us to analyze the problem~

20250718_054310.mp4
<https://drive.google.com/file/d/1Y-A-9FjwtPFQfG2IywpKNpMOQTL9knyd/view?usp=drive_web>

gary8688, 2025-07-29:
> Hello, we do not have a Google account. Please compress it into a zip file
> and upload it

I don't know how to do that. Can you not get a Google account. It is FREE!

rouyuanzi-bambu, 2025-07-29:
[image]
Hello, I’m a staff member from Bambu. The cloud drive link you provided is currently restricted, and I’m unable to access it. I’ve submitted a request for access—could you please approve it at your earliest convenience? This will help us assist you in identifying and resolving the issue more efficiently. Thank you very much for your cooperation!

gary8688, 2025-07-29:
Approved.

gary8688, 2025-07-29:
------------------------------
*From:* Bambu_Haidiye00 ***@***.***>
*Sent:* Tuesday, July 29, 2025 6:01 AM
*To:* bambulab/BambuStudio ***@***.***>
*Cc:* gary8688 ***@***.***>; Author ***@***.***>
*Subject:* Re: [bambulab/BambuStudio] Delete saved models in collection
(Issue #7568)

*Haidiye00* left a comment (bambulab/BambuStudio#7568)
<https://github.com/bambulab/BambuStudio/issues/7568#issuecomment-3131929821>

Hello, we do not have a Google account. Please compress it into a zip file
and upload it