bambulab/BambuStudio #7572: Terrible Repetitive Bambu Studio Crashing

elon-code, 2025-07-14:
### Bambu Studio Version

2.01.01.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23H2 (KB5060999)

### Additional system information

CPU: Intel Xeon W-2265 CPU @3.50GHZ 12 Core
Memory: 64 GB (4x16)
Display Card: NVIDIA Quadro RTX 5000

### Printer

Bambu H2D

### How to reproduce

This crash was simply modifying settings of print. I believe it was specifically when I changed to 100% infill.

The crash seems to appear at different times:

- During settings changes
- During slicing
- During export
- During cutting operation
- During import

### Actual results

Varies. This crash it just gave a spinning cursor then crashed nearly instantly. Sometimes it says not responding, goes full screen (no task bar showing), then crashes.

### Expected results

Crash of software. It crashes everytime I use it. Sometimes right away, sometimes after a few minutes.

### Project file & Debug log uploads

[Slidr HoneyComb.zip](https://github.com/user-attachments/files/21222633/Slidr.HoneyComb.zip)

[debug_Mon_Jul_14_14_08_25_25156.log.zip](https://github.com/user-attachments/files/21222652/debug_Mon_Jul_14_14_08_25_25156.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

rouyuanzi-bambu, 2025-07-18:
We sincerely apologize for the inconvenience caused. Based on our preliminary investigation, the issue may be related to your graphics card. A new Beta version of the software has been released, and we recommend downloading and trying it out. (Please make sure to follow the troubleshooting steps in the link below to address any graphics card-related issues before using the software):
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60
Graphics Card Troubleshooting Steps:
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing
If the issue persists, please provide the log files from the Beta version so that we can further investigate and assist you in resolving the problem.

elon-code, 2025-08-11:
Still an issue with newest public beta release v02.02.00.68. However, I tried the fix mentioned in the forum and it did fix it. It is still an issue in my opinion however. It is going to be really frustrating to show this fix to everyone at my company who has the same graphics card, and will potentially hinder my ability to buy more bambu printers for work.