bambulab/BambuStudio #7574: AMS Auto Refil needs A PRIME TOWER

LayerLabAU, 2025-07-15:
The auto refill option with the AMS is great in theory, but whenever I use it for prints that need a flawless outside finish, it always ends up with blobs on the outside walls because there is no prime tower for it to resume on. All the technology is there, it just needs some simple programming. 

Filament Runs out (Detected by AMS) - Triggers the printer to finish the current layer, then move to the poop chute for purge.
The remaining filament is purged
New filament is loaded
Nozel heated
Extruder moves to prime tower to begin printing
Extruder moves to main object for seamless transition

Easy Peasy. Make it happen plz guys!

Martin

bendman, 2025-07-15:
@LayerLabAU in the meantime would it help if you enabled the option _Quality (Advanced) > Print infill first_ ? This way your transition blob happens during infill, not on exterior finish. If it's finishing the current layer before switching to the next spool, this setting would start your next layer (the initial "prime" of the refill spool) during infill.

How do you see your suggestion working if the object didn't already require a prime tower at the layer where the auto-refill occurs? AFAIK prime towers need to be as high as the current layer which needs priming, and since it doesn't know when/if the refill will occur when the job starts, this means all jobs with auto-refill enabled will also require printing a prime tower as tall as the model, whether it is actually useful or not, which is a lot of wasted filament and time for a prime tower that is unnecessary in most jobs.

LayerLabAU, 2025-07-15:
The problem with Auto Refill is that it is completely random when it swaps the filament over. You could print infill first, or last, it doesn't matter. The swap only occurs when the filament runs out, the printer doesn't care when/where that happens, it only cares when the filament runout sensor is triggered, beginning the blobby process. 

But yeah, your ideas are great! Something along those lines... The most important thing is that it needs to resume on the tower, not the print. Make the setting easily toggled, so for those who don't need it, don't have to use it.

LayerLabAU, 2025-07-16:
Yes I understand this is not how it works, which is why I have offered a solution as to how to make it work. Nothing is impossible :) All the hardware is there, BL just needs to program a bit of If/Then coding into the firmware to tell the printer what to do and when to do it.

hotellonely, 2025-07-17:
> Yes I understand this is not how it works, which is why I have offered a solution as to how to make it work. Nothing is impossible :) All the hardware is there, BL just needs to program a bit of If/Then coding into the firmware to tell the printer what to do and when to do it.

It's not possible.

If a prime tower is there in the gcode already,  there's no way for the printer to dynamically adjust its printing path ("insert gcode") without affecting the rest of the prime tower.

atamgp, 2025-12-06:
Would it help to also purge the new filament as replacement to using the prime tower of which the gcode cannot be adjusted?

Or optimize the invoked macro which is called to make the nozzle as ready as possible to assume clean print?