bambulab/BambuStudio #7575: Extruding filament in the air

staxborn, 2025-07-15:
### Bambu Studio Version

2.5.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab H2D

### How to reproduce

Slice the file

### Actual results

Extru

[image]

de in the air

### Expected results

Not extrude in the air

### Project file & Debug log uploads

d to upload "DRY PODS TO AMS PRO 2 H2D SETUP MET HYGRO.zip" -->

### Checklist of files to include

- [ ] Log file
- [x] Project file

bendman, 2025-07-15:
@staxborn It looks like your profile file didn't upload correctly

rouyuanzi-bambu, 2025-07-18:
Hello, thank you for your feedback. As your model was not successfully uploaded, we are unable to download it and investigate the related issue. Could you please upload the model file again so that we can proceed with further analysis? Thank you for your cooperation!