bambulab/BambuStudio #7576: Add Remote Cancel for AMS 2 Pro Drying Cycle in Bambu Handy and Studio

sealyons, 2025-07-15:
As an AMS 2 Pro + P1S user, I would like the ability to CANCEL a drying cycle remotely through the Bambu Handy app or Bambu Studio.
Currently, once a drying cycle is started manually (e.g., for PETG at 65 °C for 12 hours), there is no way to cancel it except through the printer's local interface. This presents both a usability issue and a safety concern when drying sensitive filaments or cardboard-spooled materials unattended.

Since the platform already supports remote print cancel, this request aligns with existing infrastructure. Please consider adding a "Cancel Drying" button or control within the AMS interface in both the Bambu Handy app and Bambu Studio’s Device tab.

This would allow safer and more flexible use of the AMS 2 Pro and better support for users who run long-duration drying cycles across various filament types.

Thank you,
Chris Lyons
cklyons99@gmail.com

Vitorleoron, 2025-07-18:
We need this very much, and also the possibility to start drying remotely too. I would like very much this feature.

amarce, 2025-11-14:
The stop and start drying remotely (with or without making the spools rotate) is basic and extremely needed.

Thanks a lot!