bambulab/BambuStudio #7579: Filament on External Spool causes issues

garv3, 2025-07-15:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Load black Bambu Lab PLA Basic in one AMS Slot
2. Go to Device
3. Set the filament of "Ext" to a preset of a black PLA of any brand but Bambu Lab.
4. Go to Prepare
5. Sync the AMSs (Overwrite)
6. Load any model
7. Set the model filament to the external spool (i.e. 5 or 9...)
8. Slice the plate
9. Click on print plate
10. Observe that the selected filament in the print dialog is the black Bambu Lab PLA Basic filament's AMS slot and not the external spool "Ext".
11. Select "Ext" as the filament
12. Click Print

### Actual results

You will see the following warning: `Please check the following: The filament type setting of external spool is different from the filament in the slicing file, which may cause printing issues. Please click the confirm button if you still want to proceed with printing.`

### Expected results

The filament should stay selected on the print screen and when the print is being started, there should be no warning!

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21235708/log.zip)

[Projekt.zip](https://github.com/user-attachments/files/21235711/Projekt.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Qwelld, 2025-07-15:
Yes I have opened a similar bug with "Studio never matches external filament error #7577"

XinZhangBBL, 2025-08-21:
Thank you for your feedback. Please check if the problem has been resolved. https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85