bambulab/BambuStudio #7580: nozzle size

Pandygit1a, 2025-07-15:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

11

### Printer

p1s

### How to reproduce

unable to change nozzle size in bambu studio

### Actual results

printer parts can't change the nozzle had to do a workaround via orca slicer to change the nozzle size

[image]

### Expected results

still will not change it blank out

### Project file & Debug log uploads

no log s

### Checklist of files to include

- [ ] Log file
- [ ] Project file

eried, 2025-07-17:
yeah it is so stupid! on h2d there is not even how to trigger the automatic detection of nozzle.... so if you forgot you are forced to go back to the printer

Pandygit1a, 2025-07-17:
USE ORCA SLICER

mytechguyri, 2025-10-16:
Unfortunately, the asshats at Bambu keep seizing control of the printer I OWN with no justification or reasonable basis to do so.   I think I'll by an Elegoo Centauri Carbon or maybe a FlashForge AD5X for my next printer.   I'm sick and tired of Bambu telling me what I'm ALLOWED to do with MY MACHINE