bambulab/BambuStudio #7581: Continously crashing on Mac OS

michaelvango, 2025-07-15:
### Bambu Studio Version

02.01.01.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.5

### Additional system information

CPU: M1 Pro
Ram: 16GB

### Printer

Bambu Labs P1S with AMS 2 Pro

### How to reproduce

1. Launch studio
2. print somehting
3. let app sit in background for an hour or so
4. CRASH! BAM! I get crash report, I send to apple

### Actual results

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [91930]

### Expected results

Not to crash!

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21239771/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

harry-bbl, 2025-07-31:
@michaelvango If the application crashes when wake-from-sleep, please try to reproduce the issue with version 2.2 beta, and attach the crash report and log file to this issue. Thank you for your cooperation.

SteffMUC, 2025-08-04:
> Hi [@harry-bbl](https://github.com/harry-bbl),I can see the same problem on my intel based Mac, will try 2.2 beta and will come back here. Regards Stefan

With 2.2.0.68 I cannot any longer observe a crash after wake-up from sleep.