bambulab/BambuStudio #7582: BambuStudio crashes after sending gcode to A1

amethystalice23, 2025-07-15:
### Bambu Studio Version

Bambu_Studio_linux_fedora-v02.01.01.52.AppImage

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 41

### Printer

Bambu A1 & A1-mini

### How to reproduce

This started a week or two ago, I also tried downgrading to the previous release and I get the same issue.
I slice a model, send to printer, it changes to the device page, the file downloads to the printer and starts the print, bambulab immediately crashes.   If i run bambustudio again, it offers to reload the crashed workflow.  but then if i go to the device tab it will display for a second or two then seg fault again.

### Actual results

segmentation fault

### Expected results

[debug_Tue_Jul_15_18_50_01_87849.log](https://github.com/user-attachments/files/21240103/debug_Tue_Jul_15_18_50_01_87849.log)

### Project file & Debug log uploads

(gdb) where
#0  0x0000000000000000 in ??? ()
#1  0x00000000040b0595 in wx_on_channel_event ()
#2  0x00007f1a3217328c in g_main_dispatch (context=0xc2beae0) at ../glib/gmain.c:3357
#3  g_main_context_dispatch_unlocked (context=0xc2beae0) at ../glib/gmain.c:4208
#4  0x00007f1a321d38b8 in g_main_context_iterate_unlocked.isra.0 (context=0xc2beae0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/gmain.c:4273
#5  0x00007f1a32179377 in g_main_loop_run (loop=0xc86fc50) at ../glib/gmain.c:4475
#6  0x00007f1a329efb35 in gtk_main () at ../gtk/gtkmain.c:1329
#7  0x00000000040affc5 in wxGUIEventLoop::DoRun() ()
#8  0x00000000041d28ad in wxEventLoopBase::Run() ()
#9  0x000000000419ef1d in wxAppConsoleBase::OnRun() ()
#10 0x0000000004228c34 in wxEntry(int&, wchar_t**) ()
#11 0x00000000015dcf8b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#12 0x0000000000a3e2ff in Slic3r::CLI::run(int, char**) ()
#13 0x00000000009188e3 in main ()

### Checklist of files to include

- [x] Log file
- [ ] Project file

amethystalice23, 2025-07-18:
I have two printers, and if one of them is running I cannot view the Device tab at all, the moment it gets a status update from the running printer it crashes. So there is no way to switch device until both printers are idle. This is crippling now.

walterwongbbl, 2025-07-28:
@amethystalice23 
Please re-verify using the beta version 2.2, and I will follow up on this issue continuously.
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60

amethystalice23, 2025-07-28:
I already tested that beta a couple of weeks ago.  but downloaded again and retested.
./Bambu_Studio_linux_fedora-v02.02.00.60.AppImage
Open Bambu Studio, click on Device Tab.  few seconds later it crashes.
Doesn't matter if the printer is running or not, it crashes every time the page is opened.

[debug_Mon_Jul_28_09_18_01_14271.log](https://github.com/user-attachments/files/21464909/debug_Mon_Jul_28_09_18_01_14271.log)

tuxmountain, 2025-08-15:
I have exactly the same problem on flatpak Fedora 42. It use to be OK couple months ago. I try almost all version available on flatpak today from 1.9.7 to 2.1.1 but always when I open the device tab, it crash!