bambulab/BambuStudio #7587: Studio not showing correct color

ipodsync, 2025-07-16:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5

### Additional system information

M4
24G RAM

### Printer

H2D

### How to reproduce

Device Settings
Select H2D
AMS Color shows white, but filament is black
AMS Material Settings dialog shows correct black color

### Actual results

Studio shows wrong color, AMS Material Settings  shows correct color

### Expected results

Studio should show the same color asAMS Material Settings 

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-07-23:
Hello, thank you for your feedback, and we sincerely apologize for the inconvenience caused. To help us analyze the issue, please follow the steps below to provide the log files.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

joepadmiraal, 2025-11-03:
@XinZhangBambu After my print was done I reloaded the filament and after that the problem was gone.
I did make a screenshot of the filament in slot 3 when it showed the wrong color:
[image]

And this is a screenshot of the correct color:
[image]