bambulab/BambuStudio #7589: Unable to Color Change Model Parts

mjhend10, 2025-07-17:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel i5-3210M CPU 2.50GHz. 16GB ram, Intel HD Graphics 4000 32mb, 64 bit system

### Printer

Bambu A1

### How to reproduce

1. Two models on a build plate
2.  Selecting both
3.  Mouse right click 
4.  Choose 'set filament for selected items' from menu 
5.  Choose filament color 
6.  Outcome: the models don't change to the chosen color out of five different colors under project filaments

### Actual results

[image]

### Expected results

Should be able to change the selected model to a color from chosen project filaments.

### Project file & Debug log uploads

[LazyStitch_articulated_NOAMS.zip](https://github.com/user-attachments/files/21275645/LazyStitch_articulated_NOAMS.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-07-22:
Hello, I did not reproduce it
can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

rouyuanzi-bambu, 2025-07-22:
Hello, based on the model object, it appears that the issue you encountered is due to the model having already been colored using the painting tool. The software defaults to prioritizing the settings from the painting tool. If you wish to use a different method to set the filament color, please follow the steps shown in the video: first, use the "Erase all painting" function to remove the existing coloring, then right-click the model and select "Set Filament for selected items" for the changes to take effect.

[image]

mjhend10, 2025-07-22:
Hello, thanks for reaching out with the video. This makes sense. This ticket can be closed.

closed 2025-07-23 by Haidiye00