bambulab/BambuStudio #7599: Calibration Menu available for other Bambu printers but not H2D

chris99336, 2025-07-18:
### Bambu Studio Version

Beta 2.2.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu H2D

### How to reproduce

Select the H2D printer and the calibration tab disappears. Select the X1C and it reappears.

### Actual results

Select the H2D printer and the calibration tab disappears. Select the X1C and it reappears.

### Expected results

The calibration tab should be available when using the H2D

### Project file & Debug log uploads

[Bambu H2D Issue.zip](https://github.com/user-attachments/files/21309124/Bambu.H2D.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JinxCappa, 2025-08-07:
I'm also curious about this. It was added in this commit 06fa194bbe1cfdb397392158707e334c4f29de86. What was the issue that this FIX: was needed?

JinxCappa, 2025-08-31:
Is there any update to this @zhimin-zeng-bambulab? It seems that H2D owners are penalized by this specific limitation without any public reasoning. I simply would like to generate temp towers for my H2D and may a retraction test. The calibration menu just makes it easier to accomplish these tasks and the H2D gets no love in this regard. If there's a legit reason it was excluded then I am fine with that and hope we can work around it but there no information about why the decision was made anywhere.

zhimin-zeng-bambulab, 2025-09-01:
we will support in the next version

JinxCappa, 2025-09-15:
I see the timeline has changed. That's fine; I look forward to the release. In the meantime, I added the H2S printer to studio, do my calibration tests there, then switch back to the H2D. It's been fine so far.

chris99336, 2025-10-15:
> Just wanted to confirm that this is fixed in version 2.3.0 via commit [279d7f5](https://github.com/bambulab/BambuStudio/commit/279d7f5410811c41bee8df195ddf821383b46fd4). The H2D finally has the calibration menu. This issue can be closed.

I came here to confirm the same. I have been running in calibrations tonight without any workarounds. Only downside is it doesn't include the YOLO flow rate options but I will take what I can get.