bambulab/BambuStudio #7601: Adding possibility to add paragraphs to the text feature

phtevenw, 2025-07-18:
**Is your feature request related to a problem? Please describe.**
It is not posbbile to add text with paragraphs which makes it impossible to have good text features on and in parts.

**Describe the solution you'd like**
Adding the option to make more lines of text in the text feature

**Describe alternatives you've considered**
Putting multiple text feature on parts which is impossible to align in the right way. 

**Additional context**

I added two pictures one from orca slicer where it is possible to add lines.
one from bambu where i cant put  lines of text in.

[image]
[image]

phtevenw, 2025-07-18:
> https://github.com/bambulab/BambuStudio/releases Hello, please use the latest beta version. Thank you~ Hello, we have just upgraded our text tool. For Duoxing, we will consider it in the future If you currently need multiple lines, you can manually generate two lines of text

I downloaded the beta, I see that the text field is bigger, i dont see how i can make a paragraph?

phtevenw, 2025-07-18:
Would be much appreciated as it is impossible to work with multiple text features and align them in a good way.

MrMeatchunks, 2025-10-19:
This was mentioned in July, has it been updated in the beta version yet?