bambulab/BambuStudio #7602: "Recently Opened" just clean the list

ajvctrindade, 2025-07-18:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.5 (24F74)

### Printer

Bambu Lab X1C

### How to reproduce

I have no ideia. From time to time the "Recently Opened" just clean the list.

### Actual results

"Recently Opened" just clean the list

### Expected results

"Recently Opened" just clean the list

### Project file & Debug log uploads

The problem is not realted with any project. "Recently Opened" just clean the list

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rouyuanzi-bambu, 2025-07-21:
Hello, could you please provide your log file so that we can further analyze the issue?
The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

MackBambu, 2025-08-04:
Does "clean the list" mean that the list will contain nothing?

ajvctrindade, 2025-08-19:
Sorry for the delay. Yes, that's correct.