bambulab/BambuStudio #7614: missing the first layer flow setting in latest update/

Mightywiz, 2025-07-19:
### Bambu Studio Version

2.1.1.5.2

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

A1

### How to reproduce

it's this way in the latest update

### Actual results

option is now missing

### Expected results

click the advance section and scroll down to "initial layer flow"  and be able to adjust from 1 to .92 so i can get a smooth first layer.
now that it has been removed i'm kinda screwed unless you have a different way to fix it.

### Project file & Debug log uploads

nothing to show

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MadMax389, 2025-07-19:
It's there. You have to have "develop" mode enabled in preferences.

[image]

JiaYun-Bambu, 2025-07-21:
Hello, thank you for your feedback!
You can follow the helpful tip shared by other users — please try enabling Developer Mode in Preferences. Once it’s turned on, you’ll find the Initial Layer Flow option back in the Advanced section.
Thanks again to everyone for the support and suggestions! 
Let us know if you need any more help.

Jaggana69, 2026-04-22:
Thanks, that was helpfull!
Why isn't automaticly turned on for the Advanced on?
Can we remove Develop Mode check and keep Initial layer flow ratio modified?