bambulab/BambuStudio #7616: 手动绘制支撑后, 树状支撑穿模

shichao402, 2025-07-19:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

手动绘制支撑, 选择树状支撑, 有机树.

### Actual results

[image]
可以看到底部这里支撑穿透了模型

并且如果手动支撑绘制出现在了非悬空区域, 例如垂直面上. 支撑甚至会出现在这些面上.

### Expected results

我这个模型去年打印的, 今天需要调整一下. 发现按之前的绘制方法出来的支撑非常奇怪. 

### Project file & Debug log uploads

[20虹吸下水.zip](https://github.com/user-attachments/files/21328845/20.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JiaYun-Bambu, 2025-07-25:
Hello,

We will investigate the detailed cause of the support piercing issue further. In the meantime, you can try switching to Slim Tree or Hybrid Tree supports to see if that helps.

Thank you for your understanding and patience!

NothFie, 2025-08-02:
@shichao402 https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.68

shichao402, 2025-08-02:
[image]
[image]
我直接打开了之前的3mf, 修改为手动有机树, 看起来正常. 重新涂抹支撑位置也是正常的.

closed 2026-09-22 by shichao402