bambulab/BambuStudio #7637: Support Filament menu doesn't show material number and color

hotellonely, 2025-07-21:
### Bambu Studio Version

2.2.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 15.5

### Printer

Bambu Lab H2D

### How to reproduce

[image]

### Actual results

uhm see image attached above.

### Expected results

show color, filament number, filament type

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JiaYun-Bambu, 2025-07-22:
Hi, thank you for your report!
This issue has been identified and will be fixed in an upcoming version in the near future.
We appreciate your patience and understanding — sorry for the inconvenience!

hotellonely, 2025-08-12:
> <img alt="Image" width="715" height="342" src="https://private-user-images.githubusercontent.com/132528822/475834875-b6899500-6bb5-4d29-be9e-e0801fba2a27.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTUwMDc4NDcsIm5iZiI6MTc1NTAwNzU0NywicGF0aCI6Ii8xMzI1Mjg4MjIvNDc1ODM0ODc1LWI2ODk5NTAwLTZiYjUtNGQyOS1iZTllLWUwODAxZmJhMmEyNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgxMlQxNDA1NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05Y2NjYTE5ODkwMzc2M2RhOTZkMjFlMTQwZDE1ZjU3ODUwNGM5ZDA4MTk5ZDJjYWE2MzQxYTI3YjVmNGUzN2M2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6YDxN3XPqetQWjPIWCdzYh1Ix1UHAZDssMCMhhS7f3s"> Hello, we apologize for the inconvenience. We have already made internal repairs and expect the next version to be released. This is a temporary package, you can try it out https://github.com/Haidiye00/BambuStudio/actions/runs/16721463125 <img alt="Image" width="2000" height="664" src="https://private-user-images.githubusercontent.com/132528822/475834941-dfdc4f3d-b119-49fd-815f-4355820dd20e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTUwMDc4NDcsIm5iZiI6MTc1NTAwNzU0NywicGF0aCI6Ii8xMzI1Mjg4MjIvNDc1ODM0OTQxLWRmZGM0ZjNkLWIxMTktNDlmZC04MTVmLTQzNTU4MjBkZDIwZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgxMlQxNDA1NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03M2FhMDg5ZDIwYzQ4MWM3YzM5ZmJhNGUyOTA4OTY3MWUxZGNiZjU1YzNhNmQxOWEzY2Q2MDBjMTM4ODYyYzYyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.IhKcoGdHxSnj3jpnrM7F2gj2MZsXCuOqS4ybb5b6TKU">

Thanks for fixing 海底椰 :)