bambulab/BambuStudio #7638: Can't stop a print from Bambu Studio

spadrille2, 2025-07-22:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.5

### Printer

Bambu Lab H2D

### How to reproduce

Launch a print
Push the stop button on Bambu Studio

### Actual results

A window asp me if i want to stop the print? I click Yes. The window disappear and the print continues as if nothing happend.

### Expected results

The print should stop as id i did the same on Bambu Handy or the printer itself.

### Project file & Debug log uploads

[debug_Mon_Jul_21_17_44_31_36936.log.0.zip](https://github.com/user-attachments/files/21360268/debug_Mon_Jul_21_17_44_31_36936.log.0.zip)

[Rangement tête laser.3mf.zip](https://github.com/user-attachments/files/21360280/Rangement.tete.laser.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

XinZhangBBL, 2025-07-24:
Could you please attach the network log files, eg xxx_network_Wed_Jul_23_21_07_57.log

XinZhangBBL, 2025-07-24:
Could you please update to https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60, and give it a try?

spadrille2, 2025-07-24:
> XinZhangBBL
>  left a comment
> (bambulab/BambuStudio#7638)
> Could you please update to https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60, and give it a try?

Before updating i did a test with the same project and this time it worked. I already notices this behaviour before. Sometimes i can stop a print from Bambu Studio, sometimes i can’t. So, i’m not sure that trying the version you propose could help since the bug is difficult to replicate precisely.

Patrick Spadrille

XinZhangBBL, 2025-07-24:
This is mainly because the network library logs on MAC were only added in the latest version. If you encounter any problems while using the new version, you can send us the logs at any time.