bambulab/BambuStudio #7639: 上传至fluidd打印软件闪退

meatmeatghost, 2025-07-22:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

windows11

### Additional system information

i7-9750H 
32g
2080mq

### Printer

voron trident

### How to reproduce

切片
上传并打印
上传进度保持进度0%

### Actual results

上传未成功
软件闪退

### Expected results

上传未成功
软件闪退

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21361156/log.zip)

[立方体.zip](https://github.com/user-attachments/files/21361171/default.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

ShanCang-BamBu, 2025-08-01:
Hello, thank you for your feedback! Based on the 3MF file you provided, we’ve identified the cause of the crash and will fix it in an upcoming version as soon as possible.

qusheng-bambu, 2025-08-28:
We have fixed this issue in the latest version. You can verify it with the latest version. If the verification is successful, you can close this issue

https://github.com/bambulab/BambuStudio/releases/tag/v02.02.01.60