bambulab/BambuStudio #7641: Ständiger Absturz bei Slicen von Modellen

Snackboy88, 2025-07-22:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11 Pro 24H2 Betriebssystembuild: 26100.4652

### Additional system information

Dell Precision 3280 Compact, Intel(R) Core(TM) i7-14700 (2.10 GHz), 64,0 GB RAM, Nvidia T1000 8GB

### Printer

Bambu Lab P1S

### How to reproduce

Habe die Datei neu eingefärbt, und immer wenn ich slicen tue, und der Prozess fast am Ende ist, kommt das Windows Drehrad und der Bambu Lab Slicer schließt sich komplett. Bei Neustart fragt er mich ob ich das noch nicht fertig gestellte Objekt wieder bearbeiten will. Daraufhin kann ich dies tun und der zuletzt gespeicherte Fortschritt ist wieder da. 

### Actual results

Ich fange wieder an es zu slicen und wieder stürzt das Programm an der selben Stelle ab.

### Expected results

Nachdem es erfolgreich gesliced wurde, soll es an den Drucker übertragen werden.

### Project file & Debug log uploads

[crash_Tue_Jul_22_13_42_56_0_done.zip](https://github.com/user-attachments/files/21366184/crash_Tue_Jul_22_13_42_56_0_done.zip)

[Stitch+Clock+V1+Gen+PLA.zip](https://github.com/user-attachments/files/21366200/Stitch%2BClock%2BV1%2BGen%2BPLA.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JiaYun-Bambu, 2025-07-24:
Thank you for sharing the detailed information.
We see that you’re experiencing frequent crashes during slicing and sending tasks. This may be related to your NVIDIA Quadro GPU setup.
Please refer to this community thread where other users have discussed and resolved similar issues:
https://forum.bambulab.com/t/bambu-studio-crashes-after-slicing-solved-nvidia-problem/162392
Also，we have released a new version that addresses some crash-related issues. Please try downloading and using the updated software to see if it resolves your problem. Thank you for your support!
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60
If the issue still persists after trying the suggestions in that post, please let us know — we’ll arrange for the development team to check further if needed.
Thank you for your patience and support!

ShanCang-BamBu, 2025-08-13:
Hello, thank you for your feedback!
We have successfully reproduced and fixed the issue.Please try using the latest version of our Studio software. 
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85
If you encounter any other problems, feel free to reach out to us anytime. Thank you for your support!