bambulab/BambuStudio #7675: EGL Issue with latest Release and main branch

buelowp, 2025-07-26:
### Bambu Studio Version

2.1.1

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 25.04

### Additional system information

Operating System: Kubuntu 25.04
KDE Plasma Version: 6.3.4
KDE Frameworks Version: 6.12.0
Qt Version: 6.8.3
Kernel Version: 6.14.0-24-generic (64-bit)
Graphics Platform: X11
Processors: 32 × AMD Ryzen 9 9950X 16-Core Processor
Memory: 91.9 GiB of RAM
Graphics Processor 1: NVIDIA GeForce RTX 5070 Ti
Manufacturer: System76

### Printer

Bambu X1C

### How to reproduce

* Download latest
* Update dependencies
* Execute appimage

### Actual results

```
pete@theilo:~/Downloads$ ./Bambu_Studio_ubuntu-24.04_PR-7292.AppImage 
[2025-07-26 12:57:11.199901] [0x00007af66c9d2440] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/pete/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/pete/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:624867): Gtk-CRITICAL **: 12:57:11.586: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:624867): Gtk-CRITICAL **: 12:57:11.586: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:624867): Gtk-CRITICAL **: 12:57:11.586: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:624867): Gtk-CRITICAL **: 12:57:11.587: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:624867): Gtk-CRITICAL **: 12:57:11.588: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:624867): Gtk-CRITICAL **: 12:57:11.588: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Could not create GBM EGL display: EGL_SUCCESS. Aborting...
Aborted (core dumped)
```

### Expected results

App should run. I can't find results related to EGL_SUCCESS but it still causing a fault, so I haven't been able to track this down. I did build from source as well, and got the exact same result.

### Project file & Debug log uploads

No logs, the above is as far as it gets

### Checklist of files to include

- [ ] Log file
- [ ] Project file

guanyun-gudujian, 2025-08-06:
@buelowp 
currently BS do not support egl natively

we have fallback to x11 mode on wayland +nvidia platform

pls help to check if follow package can work:

ubuntu24.04: https://github.com/bambulab/BambuStudio/actions/runs/16650306777/artifacts/3659182163

thanks

ProxiBlue, 2025-08-08:
@guanyun-gudujian 

No workarounds exist. They were cleared prior to first attempt.

Proof:

```

HOST:[Fri Aug 08 12:38:56] lucas@multivac ~/Downloads $ unset __EGL_VENDOR_LIBRARY_FILENAMES
HOST:[Fri Aug 08 12:39:07] lucas@multivac ~/Downloads $ echo $__EGL_VENDOR_LIBRARY_FILENAMES

HOST:[Fri Aug 08 12:39:16] lucas@multivac ~/Downloads $ ./Bambu_Studio_ubuntu-24.04_V02.02.00.68.AppImage 
[2025-08-08 12:39:23.811418] [0x00007137ab9d9480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
after dlopen, network_module is 0x63a316b45c40

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.357: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.358: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.360: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.360: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.360: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.361: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.361: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.361: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.366: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:481428): Gtk-CRITICAL **: 12:39:24.367: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Could not create GBM EGL display: EGL_NOT_INITIALIZED. Aborting...
Aborted (core dumped)

```

ProxiBlue, 2025-08-08:
trying latest release ubuntu24.04 image: same issue

```
HOST:[Fri Aug 08 12:44:09] lucas@multivac ~/Downloads $ unzip BambuStudio_ubuntu-24.04_PR-7292.zip 
Archive:  BambuStudio_ubuntu-24.04_PR-7292.zip
  inflating: Bambu_Studio_ubuntu-24.04_PR-7292.AppImage  
HOST:[Fri Aug 08 12:44:44] lucas@multivac ~/Downloads $ chmod +x Bambu_Studio_ubuntu-24.04_PR-7292.AppImage
HOST:[Fri Aug 08 12:44:49] lucas@multivac ~/Downloads $ ./Bambu_Studio_ubuntu-24.04_PR-7292.AppImage 
[2025-08-08 12:44:59.635078] [0x00007eef83953480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/lucas/.config/BambuStudio/plugins/libbambu_networking.so
after dlopen, network_module is 0x625a3d780ff0

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.323: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.324: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.325: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.325: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.326: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.326: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.326: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.326: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.331: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:493189): Gtk-CRITICAL **: 12:45:00.331: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
Could not create GBM EGL display: EGL_NOT_INITIALIZED. Aborting...
Aborted (core dumped)

```

ProxiBlue, 2025-08-08:
@guanyun-gudujian 

I found the fix:

```
sudo ubuntu-drivers autoinstall
reboot
```

After that latest image works :)

buelowp, 2025-08-09:
Ok, so I did what was suggested above, using ubuntu-drivers, and it didn't work. I downloaded the latest from the link above, same error.

This is an AMD box, with an AMD discrete GPU and an NVIDIA 5070TI. I blacklisted the discrete AMD GPU, rebooted, and it's working now, though using a LOT of CPU.

I disabled the discrete AMD GPU using
```
pete@theilo:~/Downloads$ cat /etc/modprobe.d/blacklist-amd.conf 
blacklist radeon
blacklist amdgpu
```

Now, WebKitWebProcess is very busy and doesn't seem to be slowing down.

```
   5965 pete      20   0   70.6g 546696 170472 S  29.0   0.6   0:56.54 WebKitWebProces                                                                                            
   5856 pete      20   0   72.2g 582540 217248 S   2.3   0.6   0:10.22 bambustu_main   
```

faellacurcio, 2025-08-14:
@guanyun-gudujian 
I have a fresh install of Ubuntu 25:
- AMD Ryzen™ 5 3600 × 12
- Wayland 
- Linux 6.14.0-27-generic
- NVIDIA GeForce GTX 1660 Ti

The flatpak installation does not load the 3d rendering on the Prepare tab.

However, the appimage you shared `ubuntu24.04: https://github.com/bambulab/BambuStudio/actions/runs/16650306777/artifacts/3659182163` Seems to work just fine.