bambulab/BambuStudio #7679: [H2D] Freeze when starting a second print from USB stick without reboot

sixjeff, 2025-07-27:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.5

### Printer

Bambu Lab H2D

### How to reproduce

1. Insert a FAT32 USB stick into the Bambu Lab H2D.
2. Start a print from a G-code file on the USB stick.
3. When the print finishes, go back to the USB menu (without turning off the printer).
4. Select another file from the USB stick and start a new print.
5. The UI freezes and requires a power cycle.

### Actual results

After completing a print from a USB stick, selecting another file from the USB and starting a new print always causes the printer UI to freeze. The touchscreen becomes unresponsive and the only way to recover is to power off and on the printer.

### Expected results

The printer should be able to start a new print from the USB stick immediately after the previous print finishes, without freezing or requiring a reboot.

### Project file & Debug log uploads

Not applicable. The issue happens with any file, so there is no specific project file or log attached.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sixjeff, 2025-07-27:
This issue started suddenly. USB printing was working fine before, but now the freeze happens every time after finishing a print and trying to start another one from the USB stick.

myfooocus, 2025-08-21:
I have the same issue, so it is a bug for sure. MY whole printer has frozen now completely not sure what to do.