bambulab/BambuStudio #7683: Bambu Studio Crashing_RA_X1E_Ric

Ric-RA, 2025-07-28:
### Bambu Studio Version

v02.02.00.60-20250717193059 & v02.01.01.52-20250616155614

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	13th Gen Intel(R) Core(TM) i7-13700   2.10 GHz
Installed RAM	64.0 GB (63.7 GB usable)
System type	64-bit operating system, x64-based processor
GPU processor:		NVIDIA RTX A2000 12GB

Edition	Windows 11 Business
Version	24H2
Installed on	‎14/‎11/‎2024 | OS build	26100.4652
Experience	Windows Feature Experience Pack 1000.26100.128.0

### Printer

Bambu Lab X1E - Local LAN Network ONLY

### How to reproduce

Load STL or STEP file
Run Slice (any settings)
Bambu Studio then and always crashes

### Actual results

Bambu Studio Crashes - that is, the software simply closes down. I cannot take apicture.

### Expected results

As per normal, the software shown allow me to Print Part (that is, Send to Print)

### Project file & Debug log uploads

[PCR-24077_New Weight Bar_C04B.zip](https://github.com/user-attachments/files/21457794/PCR-24077_New.Weight.Bar_C04B.zip)

[bambu-studio.exe.3720.zip](https://github.com/user-attachments/files/21457806/bambu-studio.exe.3720.zip)

[bambu-studio.exe.9312.zip](https://github.com/user-attachments/files/21457969/bambu-studio.exe.9312.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JinGangBambu, 2025-07-28:
Hi,
Could you please provide the stl or step file on that issue has occurred?It looks that the file uploaded is a 3mf file, not stl or step file that you have mentioned。
Thank you for your understanding and support!

ShanCang-BamBu, 2025-08-12:
Hello, thank you for your feedback!
Is this issue still occurring? Have you tried using the latest version of our Studio software?
Based on our experience, issues like this can sometimes be related to NVIDIA GPUs. You might find the following forum post helpful for troubleshooting.
https://forum.bambulab.com/t/bambu-studio-crashes-after-slicing-solved-nvidia-problem/162392
If the problem persists, please provide the relevant STL and STEP files, and we will promptly try to reproduce and resolve the issue.
Thank you for your understanding and support!

Ric-RA, 2025-08-13:
Hello ShanCang, your Nvidia fix worked just fine, thank you kindly :-)

closed 2025-08-14 by ShanCang-BamBu