bambulab/BambuStudio #7685: 支撑生成最底层角度错误

Maid-Unknown, 2025-07-28:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

H2D

### How to reproduce

[image]

可以清楚的看到，在选择了直线支撑主体下。如果选择支撑顶面为同心，直线，或者网格。除了选择交叠的直线外。都会出现支撑出现的第一层（图中深绿色）的拉线方向变成和直线支撑主体同向的问题。这样直接会导致打印失败。

### Actual results

2

### Expected results

[image]

此为设置为同样的直线主体，支撑顶面设置为交叠的直线。可以看到支撑面出现的首层为正确的与直线主体呈90度。

希望能尽快在软件中修正支撑面首层角度，在选择同心，直线，网格图案下。

### Project file & Debug log uploads

4

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JinGangBambu, 2025-07-28:
Hi, 
Could you please provide the 3mf file that occurs this issue for us? Thank you for your understanding and support!

Maid-Unknown, 2025-07-30:
多测试了几下，如果选择标准方块，在底面悬垂角度特别大的情况下，就会触发这个问题。这问题好像会在悬垂角度特别大，或者完全90度悬垂中出现

[方块测试.gcode.zip](https://github.com/user-attachments/files/21505461/gcode.zip)

Maid-Unknown, 2025-07-30:
是的，大角度看起来是这个问题。但是我上面给您的那个盒子的演示，看起来就是第一层就不对的样子，因为那个盒子是完全90度水平的，十分感谢。另外希望支撑主体能提供一个和填充一样的方格款式，现在支撑主体里的那个网格，工程材料会打印失败。很大的模型又不能选择直线主体，蜂窝那个生成也不好。所以希望能有一个好用的网格款式，感谢

NothFie, 2025-07-30:
@Maid-Unknown 好的，我会反馈这一问题，但要实装这一功能可能需要一段时间。也许您可以尝试下使用混合树+闪电填充(普通支撑现在暂不支持闪电填充)？