bambulab/BambuStudio #7693: H2D Vase mode with timelapse introduces "seam like" flaw at layer boundary

Tawpie, 2025-07-29:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.5

### Printer

Bambu Lab H2D

### How to reproduce

1. create new project
2. add primitive cube at default settings, BL Basic filament
3. set to spiral vase mode
4. slice
5. print -> enable timelapse -> send

### Actual results

during printing the motion system has a slight pause and z-hop at the layer boundary, this causes what looks like a very nice seam at the extrusion discontinuity.

Loop the video file, you can observe the print bed motion
In the photo, the top is a cube with timelapse on and the bottom is the  same model with timelapse off

[image]

https://github.com/user-attachments/assets/78b2e052-cb24-491d-a577-35c29a453438

### Expected results

no seam-like features in a vase mode print done with timelapse enabled.

NOTE: if you do **not** enable timelapse, vase mode works as expected

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/user-attachments/files/21478842/Cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

psiberfunk, 2025-07-29:
P.S.  This seems to be either an MC bug or some kind of weird intermediate pause causing bulging.  If the MC firrmware can't be patched to avoid this stalling behavior that leaves creases it would be better to force timelapses off on H2D (and probably X1/P1 too) during vase mode, as is done on the A1 series where you get a warning about some issues.

zhimin-zeng-bambulab, 2025-08-13:
It has been fixed in [v02.02.00.85](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85), please have a try,

psiberfunk, 2025-08-13:
@zhimin-zeng-bambulab is this listed anywhere in the changelogs ?  I cannot see it as a listed bugfix anywhere, which leaves me a little confused .. is there a commit you can point to that fixes this specifically ?

psiberfunk, 2025-08-13:
I have searched through the commit history and cannot find any commit that purports to fix this.  I'm skeptical we have the right conclusion here since this feels like an MC firmware issue, but i'm happy to be proven wrong.. could you kindly point out where this is to be found ?  I will run another test print , but I am skeptical.

psiberfunk, 2025-08-13:
I can confirm this has **not** been fixed.  If you print the cube example here, the seam is still clearly visible, please re-open this.  @JinGangBambu

psiberfunk, 2025-08-13:
No timelapse:

[image]

Timelapse on:

[image]

zhimin-zeng-bambulab, 2025-08-13:
@psiberfunk  Thank you for your feedback, we will check if we missed it.

psiberfunk, 2025-08-13:
Thanks @zhimin-zeng-bambulab :  You may wish to consider that this is in fact a motion-controller parsing bug and the problem might actually be in the MC firmware instead of the slicer.. which.. yea.. that would suck.. but that team might need to fix it.  Similar bugs existed (and may still exist) in the X1 firmware.

BoBoYu-BamBu, 2025-12-02:
Thank you for your feedback. This issue has been fixed. Please update to the latest version of Studio for verification. Download link: https://bambulab.cn/zh-cn/download/studio

psiberfunk, 2025-12-02:
@BoBoYu-BamBu , and @zhimin-zeng-bambulab  This is clearly not fixed , please reopen .  Next time it would be great if you folks could test a claimed fix on a simple vase mode cylinder .  It’s a simple replication step print that proves the issue is still there. Myself and others above have replicated the issue multiple times , including with the latest revision.

malventano, 2025-12-03:
> Thank you for your feedback. This issue has been fixed. Please update to the latest version of Studio for verification. Download link: https://bambulab.cn/zh-cn/download/studio

This was confirmed to be a firmware issue by support several months ago. Gcode/Studio updates are not likely to correct it. I was able to correct by rolling an X1C back to the old version which supports X1Plus. H2D/S/C owners do not have this luxury, as it appears this bug was introduced with the newer firmware which also introduced the new UI, and it occurs across all models using that newer firmware base.

The seam is more pronounced at higher flows/speeds, as the stutter is more significant in those cases.

BoBoYu-BamBu, 2025-12-04:
> > [@malventano](https://github.com/malventano)您想回滚到哪个版本的 X1C？请提供具体的版本号，这将有助于我们分析问题。
> 
> 我回滚固件只是为了再次确认问题在该配置下是否仍然存在。我不需要任何回滚方面的帮助，因为我现在使用的是最新固件，并且我会在问题解决之前禁用外壳模式打印的延时功能。

Oh, I get it. Wanna tell me the firmware version number of the X1C you rolled back? It'll make it easier for us to analyze the problem.

psiberfunk, 2025-12-04:
> > [@malventano](https://github.com/malventano) What version of your X1C do you want to roll back to? Could you please provide the specific version number? This will facilitate our analysis of the problem.
> 
> I rolled back only to reconfirm the issue did not persist in that configuration. I do not require assistance to roll back, as I am now on current firmware and will disable timelapse for case mode prints until the issue is resolved in the future.

I think they meant to ask "what version did you roll back to that fixed it?"