bambulab/BambuStudio #7699: Possible Cyber Thread

2020Bytes, 2025-07-30:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1-Carbon

### How to reproduce

1. Actively printing a model, went to Device tab
2. Clicked on the Play in Bambu Studio to watch model being printed
3. AVG Web Shield secured a threat URL:TechScam. Not this is intermittant and I have only seen it twice over a period of about 2 months. This looks like Bambu Studio is trying to open a page using my default browser, firefox. Read the "Threat type" in the image from AVG Web Shield.

[image]

### Actual results

I can continue as normal.  Note that this appears to be a UDP packet, so you would normally not see it.

### Expected results

Bambu Labs should not be causing Cyber Security Treats.

### Project file & Debug log uploads

I prefer to not upload a project file as the information it contains is proprietary.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

2020Bytes, 2025-07-30:
Slight correction. Title should be "Possible Cyber Threat". The first threat was actually "Botnet:Blacklist" and occurred some time ago.

[image]

JohnSmithCYM, 2025-08-06:
I have asked for a confirmation from our team, and they confirmed that this is a false positive.
The port is used by the Liveview service to do a handshake between Bambu Studio and the printer’s camera, to allow the direct connection.
We have reported this as a false positive to Avast, to hopefully solve this incorrect report.
Thanks for feedback again.

PO3T, 2025-08-20:
These antivirus detections (multiple vendors) are most likely related to following ThreatFox entries - vendors use such IOC lists to enhance their detections:
https://threatfox.abuse.ch/ioc/1565755/
https://threatfox.abuse.ch/ioc/1563925/
https://threatfox.abuse.ch/ioc/1564254/

I reported through their 'Report False-Positive' function yesterday, hopefully this gets resolved soon.