bambulab/BambuStudio #7706: Bambu Studio crashes when loading .gcode file

rookuu, 2025-07-30:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 26

### Additional system information

Apple M3 Pro
36GB RAM

### Printer

P1S

### How to reproduce

1. Load external gcode file by opening it in Bambu Studio
2. Bambu Studio hangs on `loading Gcode`

### Actual results

[image]

### Expected results

It loads the file

### Project file & Debug log uploads

[DishwasherStatus_PLA_18m30s.gcode.zip](https://github.com/user-attachments/files/21514745/DishwasherStatus_PLA_18m30s.gcode.zip)

[debug_Wed_Jul_30_18_08_06_73652.log.0.zip](https://github.com/user-attachments/files/21514828/debug_Wed_Jul_30_18_08_06_73652.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JinGangBambu, 2025-09-02:
@will-reynolds Hello, this issue has been resolved. Please try using the latest version V2.2.1.60. If you have any problems, please feel free to let us know. Download link:https://github.com/bambulab/BambuStudio/releases

will-reynolds, 2025-09-02:
For some reason, github isn't accepting the file, even if I upload it as a .txt file, simply saying 'Failed to upload'.

BoBoYu-BamBu, 2025-12-02:
Thank you for your feedback. This issue has been fixed. Please update to the latest version of Studio for verification. Download link: https://bambulab.cn/zh-cn/download/studio

closed 2025-12-02 by BoBoYu-BamBu