bambulab/BambuStudio #7710: Bambu studio fails to run

TJKwispy, 2025-07-31:
### Bambu Studio Version

2.1.1

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora Linux (bazzite)

### Additional system information

Operating System: Bazzite 42
KDE Plasma Version: 6.4.3
KDE Frameworks Version: 6.16.0
Qt Version: 6.9.1
Kernel Version: 6.15.6-105.bazzite.fc42.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 32 × AMD Ryzen 9 5950X 16-Core Processor
Memory: 32 GiB of RAM (31.2 GiB usable)
Graphics Processor: AMD Radeon RX 7900 XTX
Manufacturer: Gigabyte Technology Co., Ltd.
Product Name: X570 AORUS ULTRA
System Version: -CF

### Printer

P1S

### How to reproduce

Install Bambu studio from flathub/github appimage
attempt to run

### Actual results

App does not start from either source. Running from the terminal produces the following:

flatpak run com.bambulab.BambuStudio
[2025-07-30 22:33:11.421371] [0x00007f9073a6c480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(process:2): GLib-GObject-CRITICAL **: 22:33:11.442: invalid (NULL) pointer instance

(process:2): GLib-GObject-CRITICAL **: 22:33:11.442: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

### Expected results

Bambu Studio opening

### Project file & Debug log uploads

no logs generated.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2025-07-31:
[image]
I installed the same system in a VM, and for now I'm unable to reproduce the issue.

TJKwispy, 2025-08-01:
SOLVED: BambuStudio.conf was corrupted on install, removing via Bazaar and re-installing did not correct the issue, but manually deleting BambuStudio.conf and running BambuStudio fixed the issue.

[debug_Thu_Jul_31_22_56_23_2.log.0.zip](https://github.com/user-attachments/files/21540963/debug_Thu_Jul_31_22_56_23_2.log.0.zip)

closed 2025-08-01 by TJKwispy