bambulab/BambuStudio #7714: Creating printers/selecting Nozzles does not work in all Languages

435jonas, 2025-07-31:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulab H2D

### How to reproduce

1 set Language to german (Problem solved by setting language to english),
2 under Printer --Create Printer-- --Drucker erstellen--
3 select "cant find my model" and set "a" and "b" for producer and model (any string)
4 select any Düsendurchmesser/nozzle diameter or even type in a nozzle diameter
5 click OK

[image]

[image]

[image]

[image]

### Actual results

any selected or typed nozzle diameter is invalid.
Works in english not in german.

[image]

### Expected results

[image]

### Project file & Debug log uploads

This is not projectspecific

### Checklist of files to include

- [ ] Log file
- [ ] Project file

murongguao, 2025-08-04:
We will fix the issue as soon as possible.

DanBao-Bambu, 2025-08-04:
We will fix the issue as soon as possible.