bambulab/BambuStudio #7718: Mac version continues crashing when idle (Mac M4 Max ARM64)

ppwaskie, 2025-07-31:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5

### Additional system information

Apple M4 Max 64GB of unified RAM, MBP 14".

### Printer

Bambu Labs X1C and Bambu Labs A1

### How to reproduce

I leave the Macbook powered on and on AC power, lock the screen, and walk away.  I have a pretty good chance when I come back later (couple of hours) that Bambu Studio crashed and is no longer open.  I can fire it up immediately and it detects there was an open project file before, and opens it back up.

### Actual results

App crashes.  No crashlog indicator.

### Expected results

The app doesn't crash.

### Project file & Debug log uploads

It's not specific to any project file at all.  I just leave the application open when my Macbook locks the screen, and it will eventually crash.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JinGangBambu, 2025-08-01:
Hello,
We have released a new version that addresses some crash-related issues. Please try downloading and using the updated software to see if it resolves your problem. Thank you for your support!
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.68
If the issue still persists after upgrade, please let us know — we’ll arrange for the development team to check further if needed.
Thank you for your patience and support!

ppwaskie, 2025-08-01:
Thanks, I'll give it a try.  The issue is I don't know what event causes it to segfault.  I've always used Bambu Studio on my x86 Windows laptop, but recently upgraded to my M4 Max Macbook.  I've never had a stability issue on the x86 Windows platform (I've had stability issues with x86 under Linux for sure).

If the beta also has similar issues, I'll try and get a crash dump report from MacOS and post it.

ppwaskie, 2025-08-01:
And it just crashed...

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [80773]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               02.01.01.52 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-07-31 09:53:41.4769 -0700
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        EAA9022D-AD6E-4909-7E70-EE81FF749ADD

Sleep/Wake UUID:       1BA34416-3023-4E93-979A-1FD640195124

Time Awake Since Boot: 130000 seconds
Time Since Wake:       1 seconds

System Integrity Protection: enabled

Crashed Thread:        34

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [80773]

VM Region Info: 0 is not in any region.  Bytes before following region: 4369858560
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT (graphics)           10476c000-108950000    [ 65.9M] r-x/r-x SM=COW  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x105ae8658 Slic3r::MachineObject::parse_json(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool) + 77812
1   BambuStudio                   	       0x10567e2cc wxAsyncMethodCallEventFunctor<Slic3r::GUI::GUI_App::init_networking_callbacks()::$_22::operator()(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) const::'lambda'()>::Execute() + 284
2   BambuStudio                   	       0x107c617d0 wxEvtHandler::ProcessEventLocally(wxEvent&) + 172
3   BambuStudio                   	       0x107c615a0 wxEvtHandler::ProcessEvent(wxEvent&) + 56
4   BambuStudio                   	       0x107c613f0 wxEvtHandler::ProcessPendingEvents() + 236
5   BambuStudio                   	       0x107bc56dc wxAppConsoleBase::ProcessPendingEvents() + 112
6   BambuStudio                   	       0x107c6a2ac wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
7   CoreFoundation                	       0x196a05098 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
8   CoreFoundation                	       0x196a04f80 __CFRunLoopDoObservers + 536
9   CoreFoundation                	       0x196a045f0 __CFRunLoopRun + 784
10  CoreFoundation                	       0x196a03c58 CFRunLoopRunSpecific + 572
11  HIToolbox                     	       0x1a249827c RunCurrentEventLoopInMode + 324
12  HIToolbox                     	       0x1a249b31c ReceiveNextEventCommon + 216
13  HIToolbox                     	       0x1a2626484 _BlockUntilNextEventMatchingListInModeWithFilter + 76
14  AppKit                        	       0x19a92bab4 _DPSNextEvent + 684
15  AppKit                        	       0x19b2ca5b0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
16  AppKit                        	       0x19a91ec64 -[NSApplication run] + 480
17  BambuStudio                   	       0x107b91d10 wxGUIEventLoop::OSXDoRun() + 112
18  BambuStudio                   	       0x107c6a990 wxCFEventLoop::DoRun() + 40
19  BambuStudio                   	       0x107be9f70 wxEventLoopBase::Run() + 132
20  BambuStudio                   	       0x107bc4f70 wxAppConsoleBase::MainLoop() + 120
21  BambuStudio                   	       0x107b567a4 wxApp::OnRun() + 36
22  BambuStudio                   	       0x107c17d7c wxEntry(int&, wchar_t**) + 148
23  BambuStudio                   	       0x105634638 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 468
24  BambuStudio                   	       0x104784a00 Slic3r::CLI::run(int, char**) + 33868
25  BambuStudio                   	       0x1047f20bc main + 172
26  dyld                          	       0x19657ab98 start + 6076
[... 614 lines cut ...]
      "deploymentId" : 240000005
    }
  ],
  "experiments" : [

  ]
}
}

Model: Mac16,6, BootROM 11881.121.1, proc 16:12:4 processors, 64 GB, SMC 
Graphics: Apple M4 Max, Apple M4 Max, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Mar 22 2025 02:17:00 version 23.40.30.0.41.51.180 FWID 01-989cd4a9
IO80211_driverkit-1475.39 "IO80211_driverkit-1475.39" Apr 18 2025 20:10:40
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

mhwunder, 2025-08-02:
I came here to post the same issue on a 2017 Intel iMac.  Random idle crashes.  I have not updated yet but based on the previous post it seems it doesn't help.  Here is my crash report.

[BL Crash Mac.odt](https://github.com/user-attachments/files/21559629/BL.Crash.Mac.odt)

peternlewis, 2025-08-06:
I believe it crashes for me if it loses network connection.

Using 2.1.1.52 - which is the latest according to Check For Updates.

XinZhangBBL, 2025-08-06:
Thankl you for your feedback. The problem has been fixed since https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60. The public version will be released soon

[image]

ppwaskie, 2025-08-13:
So far the new public release has been solid on my Mac.  Thank you guys!  I'm going to close this out.

closed 2025-08-13 by ppwaskie